sailoog/openplotter #219: OP forbid multiple USB WiFi clients dongle

doxav, 2018-08-22:
In a setup with the Pi3 builtin WiFi interface and >= 2 USB WiFi adapters used to connect multiple clients, OP will either show only 1 WiFi client interface, either automatically disconnect any new connection with more than 1 adapter.

e-sailing, 2018-09-15:
If you set the network mode to "client" it uses standard raspbian network setup. You can connect as many WiFi interfaces as raspbian likes.
The other network modes are made for special configurations(AP). You can build your own setup and copy it into the network folder.
The Linux  router matrix defines the sequence for the Internet gateway. You can connect to many APs but Linux will only communicate with one to the Internet. I can't see any advantage to have more than one connection to the WiFi Internet. 
Please explain what you want to achieve.

FredericGuilbault, 2019-02-13:
Does setting openplotter to "client" and use the linux way to setup your configs solve the problem for you @doxav  ?