sailoog/openplotter #224: Can SDR AIS receive on both channels A and B simultaneously?

marktolson, 2018-09-07:
In the SDR Receiver options there are 2 checkboxes (under Enable AIS reception) labelled Channel A and Channel B. You can only select one at a time. Does this mean AIS is only listening to one channel at a time or are those checkboxes purely for viewing the waterfall output when you press 'Take a look'? Is dual channel AIS possible?

e-sailing, 2018-09-15:
Yes only one channel. Yes dual channel AIS is possible. Look at the fork of RooieDirk or use the moitessier hat or other hardware.

sailoog, 2018-09-16:
we will try to implement 2 channels simultaneously just before going to beta state

marktolson, 2018-09-16:
> we will try to implement 2 channels simultaneously just before going to beta state

Awesome thanks for the hard work!