sailoog/openplotter #228: Can't connect over Wi-FI using SSH or RDP

Phil-Taylor-IW, 2018-12-10:
Hi there, I just booted openplotter_v1.0.0_img off the SD card in my Pi 3 Model B+

I can connect to the SSID: openplotter, get an IP address and ping 10.10.10.1 OK but a port scan shows no open ports and tcp/22 SSH or RDP from the MacBook fail to connect.

Looks like a great product so could someone please assist?

Thanks, Phil

andyrbarrow, 2018-12-10:
Have you checked the Raspberry Pi configuration, or from a terminal session, raspi-config. Connect via VNC and check that SSH is enabled.

Phil-Taylor-IW, 2018-12-10:
I’m in now via VNC Viewer and can confirm the SSH server is disabled so perhaps you need to update your documentation?

Thanks

Phil-Taylor-IW, 2018-12-10:
It's disabled but your documentation say that VNC, RDP and SSH are the options?

sailoog, 2018-12-10:
You are reading documentation for v0.1.x
https://sailoog.gitbooks.io/openplotter-documentation/content/en/

You need to follow docs for v1.x.x
https://docs.sailoog.com/openplotter-v1-x-x/

closed 2018-12-10 by sailoog