sailoog/openplotter #229: signalk-server-node error in 1.2 upgrade - openplotter wont load

thezerocool, 2019-01-02:
after running the GUI update tool in openplotter the system no longer functions after a reboot.  Update was from early version.

New software points to error in finding signalk-server-node folder in /home/pi/.config.  Error caused by check_vessel_self.py line 34.

Folder does not exist, it seems the upgrade package did not ensure signalk was installed despite its dependancy.

[bit.ly/2s3ZoOg](url)

sailoog, 2019-01-04:
I have just updated from OP NOOBS 1.x.x to 1.2.0 and everything is working right. You may have a failed update or a corrupted SD.

FredericGuilbault, 2019-02-13:
@thezerocool  Can you reproduce the problem in the 1.2.X ?

thezerocool, 2019-02-13:
I'll see if I can find the original image and try another upgrade.  I would have agreed with the corrupt SD if there were any other issues but this was a number of files missing.  I'm pretty sure the upgrade script just didn't identify that signalk was missing and continued regardless.