sailoog/openplotter #230: current NOOBs install cannot see wifi interface on RPi 3B

thezerocool, 2019-01-03:
"No wireless interfaces found" in Wifi GUI tool on the current version of noobs with a fresh install.  Known working wifi on the previous install with a Raspberry Pi 3B.  Changing network preferences in openplotter makes no difference.

NOOBS version: 2.8.1
Release date: 2018-04-24
Based on Raspbian (Debian Stretch)
Release date: 2018-04-18
Default login: pi / raspberry
Kernel version: 4.14
Download: https://archive.org/details/openplotter_v1.0.0_noobs
Size:   1.1G
SHA-1: 751e55c82a9bc30c66895dfe73a61248daddcc86

I will test the IMG version next.

thezerocool, 2019-01-04:
to update, it seems the IMG didnt have the same problem, wifi mode was easily set to client and wifi points appeared fine.

sailoog, 2019-01-04:
I have just tested a fresh NOOBS install and wifi is working right for me. On first run the access point is set ok. If I set "client" in Network tab, I can connect to a router without problem after reset.

dlparker, 2019-03-12:
I am also getting this problem. The access point is working properly on wlan1, but I get the "no wireless interfaces found" on mouse over of the network applet at the top of the screen.

When I try sudo ifup wlan0, I get this in /var/log/system:

`Mar 12 13:38:28 openplotter wpa_supplicant[1493]: Successfully initialized wpa_supplicant
Mar 12 13:38:28 openplotter wpa_supplicant[1493]: Could not read interface wlan0 flags: No such device
Mar 12 13:38:28 openplotter wpa_supplicant[1493]: nl80211: Driver does not support authentication/association or connect commands
Mar 12 13:38:28 openplotter wpa_supplicant[1493]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mar 12 13:38:28 openplotter wpa_supplicant[1493]: Could not read interface wlan0 flags: No such device
Mar 12 13:38:28 openplotter wpa_supplicant[1493]: rfkill: Cannot get wiphy information
Mar 12 13:38:28 openplotter wpa_supplicant[1493]: Could not read interface wlan0 flags: No such device
Mar 12 13:38:28 openplotter wpa_supplicant[1493]: WEXT: Could not set interface 'wlan0' UP
Mar 12 13:38:28 openplotter wpa_supplicant[1493]: wlan0: Failed to initialize driver interface
`
Kinda stumped at this point, will have to keep searching. This AP + client thing is cool, but new, so bugs would not be surprising.

youleaf, 2019-06-11:
up. 
with a fresh installation (raspberry 3b+, openplotter noobs image) I can only get the client mode working; in client + ap mode, only the ap works on wlan1 and wlan0 won't go up at startup nor with ifup

As a side note, in client mode  update to 1.2.0 beta fails when upgrading hostapd:

    ...
    Setting up hostapd (2:2.6-21~bpo9~rpt1) ...
    Job for hostapd.service failed because the control process exited with error code.
    See "systemctl status hostapd.service" and "journalctl -xe" for details.
    invoke-rc.d: initscript hostapd, action "restart" failed.
    â hostapd.service - Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
       Loaded: loaded (/lib/systemd/system/hostapd.service; disabled; vendor preset: enabled)
       Active: activating (auto-restart) (Result: exit-code) since Tue 2019-06-11 16:13:38 UTC; 29ms ago
      Process: 30442 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid -B $DAEMON_OPTS ${DAEMON_CONF} (code=exited, status=1/FAILURE)
    
    Jun 11 16:13:38 openplotter systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
    Jun 11 16:13:38 openplotter systemd[1]: hostapd.service: Unit entered failed state.
    Jun 11 16:13:38 openplotter systemd[1]: hostapd.service: Failed with result 'exit-code'.
    dpkg: error processing package hostapd (--configure):
     subprocess installed post-installation script returned error exit status 1
    Errors were encountered while processing:
     hostapd
    E: Sub-process /usr/bin/dpkg returned an error code (1)
    
    #### ERROR. ABORTING, PRESS ENTER TO EXIT ####