sailoog/openplotter #232: Unable to access to internet with USB Wifi dongle randomly

pkernevez, 2019-01-16:
Hi all,

I have a Wifi USB dongle. During the first boots I don't have issues, I was able to connect to internet and update openplotter.
But now I lost the capacity to connect to internet. This is due to the definition of 2 default IP routes.
Now the interface that win is not the internet interface.

My model : Raspberry Pi 3 Model B Rev 1.2
My configuration:
wlan 0 : my dongle
wlan 1 : board Wifi
Openplotter : 1.2.0 alpha

My Wifi configuration :
[image]

My routes just after the boot:
```
$ ip route show
default via 10.10.10.1 dev wlan1 src 10.10.10.1 metric 303
default via 192.168.1.1 dev wlan0 src 192.168.1.131 metric 304
10.10.10.0/24 dev wlan1 proto kernel scope link src 10.10.10.1 metric 303
192.168.1.0/24 dev wlan0 proto kernel scope link src 192.168.1.131 metric 304
```

My pb and the solution:
```
pi@openplotter:~ $ ping google.com
ping: google.com: Temporary failure in name resolution
pi@openplotter:~ $ sudo ip route delete default via 10.10.10.1 dev wlan1 src 10.10.10.1
pi@openplotter:~ $ ping google.com
PING google.com (216.58.215.238) 56(84) bytes of data.
64 bytes from zrh11s02-in-f14.1e100.net (216.58.215.238): icmp_seq=1 ttl=53 time=15.6 ms
64 bytes from zrh11s02-in-f14.1e100.net (216.58.215.238): icmp_seq=2 ttl=53 time=21.7 ms
```
I don't know why with the configuration : AP + Client the access point has a default route.

Thank to the team for this beautiful tool !
Philippe

pkernevez, 2019-01-16:
I there a public forum somewhere to ask for question before creating an issue ?

richardmbell, 2019-01-17:
Absolutely!  We are at forum.OpenMarine.net

Sent from my Verizon, Samsung Galaxy smartphone

pkernevez, 2019-01-20:
After some new tests. It seems that the default routes are applied in a random order. 
Half of the time the first default route is the route of the AP (wlan1) instead the internet client (wlan0).
An idea of the root cause ?

FredericGuilbault, 2019-02-13:
Udev node attribution is first up first attributed.  Relying on wlanX is not a stable way to do it. 

Writing udev rules that use manifacturer+ model version or MAC address would be the proper way fingerprint your devices. 

This said, IDK how openplotter do it. It's just my 2 cents on the topic.