sailoog/openplotter #234: Installation location

FredericGuilbault, 2019-01-19:
Default installation location for openplotter is `~/.config`
But it should be `~/.local/share/`

Only configuration files be in `.config` 

On a larger view. openplotter should not be in any ~/ folder. but that's a bigger job.

sailoog, 2019-02-20:
I do not like that place either. What about /usr/share?
I do not think is such big job and v2.x.x is the right moment to do it.

FredericGuilbault, 2019-02-22:
`/usr/share` is for files from packages that are not ARM specific. (Like doc, icons, and text ...)

I would check what pip have to offer in term of managed installation process (witch i have no idea, i just know that what python people do ;p ).  

Or just put your `openplotter` folder in `/opt/` as it's the place for standalone applications that don't integrate with the linux structure. (that's where I plan to put openplotter on my OS)

---

If you choose to integrate in the Linux architecture, you kinda have to do spread your app across the filesystem. Binaries in `bin`, libraries` lib` or `include`, configs in `etc` logs in /var/log`, static files in `/share`, pid and socket files in `/var/run` .... 

But in your case most will be in `/usr/local/*`.  
 - `/usr` cuz user level (runlevel3) stuff 
 - `local` cuz it's not meant to be updated by any external process like `apt` or `puppet`.

`/usr/local/bin`
`/usr/local/share`
`/usr/local/lib`
`...`