sailoog/openplotter #237: New Chart Source Tab Unusable

Yeah-Brah, 2019-01-22:
When attempting to download charts using a Pi 3 B+ with a 7" 800x480 screen the New chart source tab does not fully open. I have tried installing new drivers for the display as well as manually editing the config file but cant get it to open properly. This is the only tab I have found thus far with this issue. Attached is a picture showing the Tab as it opens now.
[image]

sailoog, 2019-01-22:
This is an opencpn issue, please report or wait for opencpn 5

FredericGuilbault, 2019-02-13:
Openplotter only packaging the opencpn application. 
As this problem append inside opencpn, it should be reported to the opencpn team.  

Closing ?