sailoog/openplotter #241: Check signalk password in security.json file

FredericGuilbault, 2019-02-13:
in the `startup` file, openplotter check in the `security.json` file for default password. But this file can be missing in 2 case,  
- Frist run. if the user have a fresh install of signalK and didin't set a username/password for the admin. This file don't exist yet. 

- Password reset. The procedure describe by the signalk project to make an admin password reset it to delete this file and restart the signalk server. 

On these 2 case, openplotter crash. 

A "Plz create an admin account" warning could be displayed if the file is not found.

sailoog, 2019-02-13:
on v2.x.x there is no password set by default. Users will set their password first time they run SK server, so checking security.json in startup has been removed.

FredericGuilbault, 2019-02-14:
Cool, so it will be fixed in 2.x.x

How experimental is the v2.x.x atm ?

closed 2019-02-14 by FredericGuilbault