sailoog/openplotter #242: util_process_exist() function don't see signalk if it been started with the -c argument

FredericGuilbault, 2019-02-13:
_I really feel like spamming the bug tracker...._

In the `startup` file at line `426`  in the function `util_process_exist`
openplotter don't detect the signalk process if it been started with the `-c` argument. 
Causing it to start a new signalk process that complain to be unable to use the port cuz the other process is already listening on it. 

This is what it's look when both process are running: 
```
pi@lysmarine:~ $ ps aux | grep signalk
pi        7354  0.0  0.0   1900   392 ?        Ss   14:15   0:00 /bin/sh /home/pi/.signalk/signalk-server
pi        7356  0.3  5.8 201476 55852 ?    Sl    14:15   0:09 node /usr/lib/node_modules/signalk-server/bin/signalk-server -c /home/pi/.signalk
```

Then if i do the same check as openplotter do for the 2 process: 
```
pi@lysmarine:~ $ cat /proc/7354/cmdline 
/bin/sh/home/pi/.signalk/signalk-server
pi@lysmarine:~ $ cat /proc/7356/cmdline 
node/usr/lib/node_modules/signalk-server/bin/signalk-server-c/home/pi/.signalk
```

e-sailing, 2019-02-13:
Startup only stops and starts the signalk process with 
"
Stop the daemon temporarily with;

$ sudo systemctl stop signalk.service
$ sudo systemctl stop signalk.socket

Start the deamon with;

$ sudo systemctl start signalk.socket
$ sudo systemctl start signalk.service
"
What's wrong? We have the two processes every time.
Signalk should be installed the way it is done in this description "https://github.com/SignalK/signalk-server-node/blob/master/raspberry_pi_installation.md" Step 3

FredericGuilbault, 2019-02-22:
```
def util_process_exist(process_name):
	pids = [pid for pid in os.listdir('/proc') if pid.isdigit()]
	exist = False
	for pid in pids:
		try:
			if process_name in open(os.path.join('/proc', pid, 'cmdline'), 'rb').read():
				exist = True
		except IOError:  # proc has already terminated
			continue
return exist 
```

As I understand it, this peice of code look for signalk-server by some kind of string comparaison of the command line used to launch the process.

The `signalk.service` file refer to user specific `ExecStart` 
```
ExecStart=/home/pi/.signalk/signalk-server 
```
Where `/home/pi/.signalk/signalk-server` file looks like :
```
#!/bin/sh
/usr/lib/node_modules/signalk-server/bin/signalk-server -c /home/pi/.signalk $*
```
Hence the `-c` in the process name. 

But I might be wrong with my signalk install as I haven't been able to follow the step3 of signalk documentation as it's an interactive process and can't be run in pi-gen. I had to read the install script file and reproduce every step in a static way. Ill do my homework and double check if have respected the signalk procedure.

FredericGuilbault, 2019-03-07:
Here, line 230 it's part of the install script:

https://github.com/SignalK/signalk-server-node/blob/30d918adfd7c9ce9c5956669ba14a85c5b197bc4/bin/signalk-server-setup#L230

sailoog, 2019-03-07:
I can not reproduce your issue. Anyway I think the correct way of checking SK status should be running systemctl status. I will change this on v2.x.x.