sailoog/openplotter #243: 0.10.0 offers update to beta, which tries 1.2.0 alpha, which fails

dlparker, 2019-03-12:
When choosing "Update OpenPlotter Beta" from updates menu, the process encounters errors not being able to access files. For example:

 chmod: cannot access 'openplotter_tmp/openplotter': No such file or directory

I believe that the problem is that the update script in 0.10.0 calls the updated_settings script in the new code (after git clone and backup):

cd ~/.config
source openplotter_tmp/update/update_settings.sh

But this script expects to be run from the update_Openplotter.sh script and to have the variable op_folder set, so when it does a cd to that variable it goes to the wrong place (home directory?).

I know that getting a new image is preferable, but I want to try this using a new install of 010.0 so that I could test the update path for an image I have that has been in use for a while and has extra stuff I installed. 

Will try updating to stable, then to beta to see if that works and update comments here.

dlparker, 2019-03-12:
Looks like a similar problem updating to 0.17.0

CREATING OPENPLOTTER CODE BACKUP...

[Errno 2] No such file or directory: '/home/pi/.openplotter/openplotter.conf'

APPLYING SETTINGS...

Parameter not found: op_folder

REMOVING UNUSED PACKAGES...

dlparker, 2019-03-12:
So, basically, I think my 0.10.0 image is orphaned. Don't know if anyone else cares, so if you decide to close this as won't fix, then I won't complain. 

Though I do think it is possible that a supplemental script posted on this ticket might be able to provide at least an upgrade 0.17.0.

jamesewilliamson, 2019-03-20:
" So, basically, I think my 0.10.0 image is orphaned"
correct. major updates require new img burn due to kernel updates and such.
no upgrades from 0.17.0 need's new img.
 http://sailoog.com/blog-categories/openplotter-rpi
v2.0.0 is in the works so if version schema is the same a new img will be released