sailoog/openplotter #259: pre-detect raspberry model when selecting Network option

FredericGuilbault, 2019-04-06:
Raspberry model can be detected with : 
`cat /proc/device-tree/model` This could help make the selection in the network table. 

Maybe in a first time by just displaying it. over the selection box. if this reveal to be relyable, to limit the selections  to the specific raspberry used. 

Would you take a PR regarding this ?

sailoog, 2019-04-07:
I like this. 
@e-sailing?

e-sailing, 2019-04-17:
It was already integrated with /sys/firmware/devicetree/base/model.
You can test it. I will make a pull request. 
(Network is a never ending story)

sailoog, 2019-04-17:
closed by https://github.com/sailoog/openplotter/pull/260

closed 2019-04-17 by sailoog

FredericGuilbault, 2019-04-18:
hummm. I feel like you guys are not developing on github and have a private code base somewhere ?

sailoog, 2019-04-18:
What do you mean? we develop in our computers then we push to our personal github account and finally we make a pull request to the active branch in sailoog repo in my case obviously is the same repo. Just like everyone does. You know this is open source and always will be, why should we hide anything? Actually we need as much help as we can get.

FredericGuilbault, 2019-04-19:
I just had impression that you where further in your work then what was on github. So I asked to avoid assumption or misunderstanding. Nowit's been clarified to me im all good.

zipperten, 2019-05-10:
Thank you for your work, i think this is the best thing that hapen foss marine world, im a old linux user that you save a lot of time, i get my new SDR AIS up in moments, instead of days! 
Thank you keep up the good work, I will buy your rpi hat when my old gps die!