sailoog/openplotter #273: Fails to install some apps

pcav, 2019-12-20:
I successfully installed 2.0.18-beta in my Debian Sid box. All fine, I update sources, installed settings and docs, but when I ask to install OpenCPN or tools panel nothing happens.
Sorry if this is misplaced, please let me know if I should write to the forum instead.
Thanks.

pcav, 2019-12-20:
After that the application hungs, and I have to manually kill it.

sailoog, 2019-12-20:
This openplotter repo is dead. You should use forum.openmarine.net or this repo: https://github.com/openplotter/openplotter-settings

Open a terminal, run "openplotter-settings", play with features and report any error in terminal.

We have never tested with Sid, just stable versions.

closed 2019-12-20 by sailoog

pcav, 2019-12-20:
This is what I did. I'll report in the other repo.
Thanks.